antodom/pwm_lib #2: installation

ranibourbara, 2018-11-14:
Dear antodom,

Can you please give a walkthrough on how to compile and stuff for using the librabry? I have never done such a process and I appear not to understand it well.

Thanks

ranibourbara, 2018-11-14:
when I just add the libraries normally without cmake compilation I get the following error:

In file included from C:\Program Files (x86)\Arduino\libraries\tc_lib-master/tc_lib.h:39:0,

                 from C:\Program Files (x86)\Arduino\libraries\pwm_lib-master\examples\basic_test\basic_test.ino:36:

c:\programdata\matlab\supportpackages\r2016b\3p.instrset\arduinoide.instrset\arduinosam-1.6.7\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\c++\4.8.3\cstdint:38:28: fatal error: bits/c++config.h: No such file or directory

 #include <bits/c++config.h>

                            ^

compilation terminated.

antodom, 2018-11-14:
Dear ranibourbara,

For using the library just add the library to the directory "libraries" in the root directory where your Arduino IDE store your projects. Usually it is directory "<your-user-directory>/Arduino/libraries". Doing that you should compile without errors any of the examples that come with the library. Evidently you must select Arduino DUE as target platform.

I hope it helps.
Best.

ranibourbara, 2018-11-14:
I am happy to hear this is simple. I am however getting the above error when running the basic_test. do you have any direction of what is the problem?

Thank you very muuch

antodom, 2018-11-14:
Hi again ranibourbara,

I would say that your library directory should be named "pwm_lib" instead of "pwm_lib-master", the directory must have exactly the same name that the library when you compile through Arduino IDE.

Please, try and tell me.
I hope it helps.

closed 2018-11-25 by antodom