antodom/pwm_lib #4: Frequency sweep glitches

Uup115, 2019-02-27:
Hello Antonio,

Very nice library.  Thanks for your contributions to the Arduino Community!

I have a problem...  My use case requires a frequency sweep from 10 Hz to 1000 Hz to 10 Hz, with 50% duty.  Step size is adjustable.  The frequency steps sometimes do not transition smoothly.  Meaning, the output will temporarily stick in the high or low state, before stabilizing at the new frequency.  Disabling interrupts, before writing the period and duty, doesn't seem to help.

Unfortunately, this is causing problems with connected device(s).  Have you encountered this problem?  Is there a fix for it?  Any assistance is appreciated.  Thanks!

Best Regards

antodom, 2019-02-27:
Hi there,

I would need more info to be of any help. It seems to me that you are changing the PWM frequencies too fast, at least for the devices you want to control, unless I am understanding incorrectly.

Just give more details, or attach code exemplifying the problem.
Best.

Uup115, 2019-02-27:
Hello Antonio,

Thanks for your quick response.

I'm updating the PWM frequency, based on the RPM of an engine.  The following code checks for an updated RPM, then sets the PWM.  The RPM value is refreshed every ~15ms.  It's unlikely the RPM will change at every refresh.  For this reason, I only set a new PWM frequency when the RPM changes. 

> if (rpm != rpmPrevious) {
        period = 0.5f + (M * (1 / ((rpm / 60) * ppr)));
        if (!pwm_pin35.set_period_and_duty(period,(period >> 1))) {
          Serial.print("[ERROR] failed to set period and duty cycle:  ");
          delay (5000);
        }
    }
> 

Worst case, is a ~15ms cycle time too fast for setting PWM?  Is there a way to better manage this?  Thanks for your assistance.

Best Regards

antodom, 2019-03-01:
Hi again,

Which are the typical PWM periods you are commanding?. Take into account that if you change the PWM period each 15 msecs. you should not be commanded PWM periods bigger than that to have a "smooth" behavior.

Best.

Uup115, 2019-03-01:
Hello Antonio,

Periods can be up to 100 msec (10 Hz).  I've tried ALL of the following:

**Increasing the RPM refresh rate to above 100 msec
+
Increasing the frequency multiplier (ppr), as shown above
+
Disabling interrupts, prior to writing the period**

It seems there is something else causing the issue.

Best Regards

antodom, 2019-03-01:
Hi there,

In order to try to clarify what happens, I would suggest you to try to find out at which frequencies you have the problem. At first sight, I think maybe the problem is that this happens when a frequency is established with provokes the PWM module to change its internal clock source. For have some certainty, it is necessary to determine at which frequencies the problem appears. A oscilloscope would be also of great help.

Best.

Uup115, 2019-03-05:
Hello Antonio,

Yes, using a scope is very helpful.  This is how I found the glitch seven days ago.  Actually, your insight has been very helpful.  You are correct.  Changing the clock source, seems to cause the glitch.  I simply setup the PWM manually.  Keeping the clock source constant, I see smooth transitions throughout the desired sweep range.  

Does your library allow for a constant clock source?  If not, this would be a great feature to add, as I would really like to keep using your library.  Thanks again for your assistance.

Best Regards

antodom, 2019-03-05:
Hi again Uup115,

In fact, it is easy to keep the same clock using pwm_lib. For that you should keep the period you command in a specific interval. To know the interval for each possible clock, you can use pwm_lib function pwm::core::max_period(uint32_t two_power). Parameter two_power specify the clock. More precisely, the 2 power prescaler applied to the system clock for determining the PWM period. In other words, this parameter defines the clock used by the PWM module. It can take values from 0 (no prescaler), 1 (prescaler of 2), 2 (prescarler of 4), ... to 10 (prescaler of 1024). Using this function you can determine at which interval of periods, when you change the PWM period the clock keeps being the same.

Another option is to take some preventive actions in advance before changing to a PWM period you know will change the internal PWM clock. Those actions can be to put the device you are controlling in a safety state during the period change. Take into account that pwm_lib does this for you when necessary when you change the PWM period using member function set_period_and_duty(), you can have a look to its code for further clarification.

I hope it helps.

Uup115, 2019-03-06:
Hello Antonio,

Thanks for your quick response.  

Regarding the first option...  The periods I need, range from 1ms to 100 ms.  Is this what you mean by interval?

From your explanation, I understand...  
 pwm::core::max_period(uint32_t two_power) sets the clock prescaler.  This means the clock won't change during interval operation, correct?  So, I should call this function first, yes?

Best Regards

Uup115, 2019-03-07:
Hello Antonio,

With your library, I don't see any way to set a constant clock.  So, I made a some modifications to "pwm_lib.h" to accommodate.  Specifically, I made changes to the following two functions (see attached file **"pwm_lib.zip"**):

**bool start (uint32_t period, uint32_t duty)**

and 

**bool set_period_and_duty (uint32_t period, uint32_t duty)**

After finding the desired clock divisor, with function pwm::core::max_period(uint32_t two_power), I simply pass it as a parameter to the modified functions...

**bool start (uint32_t period, uint32_t duty, uint32_t clkDiv)**

and 

**bool set_period_and_duty (uint32_t period, uint32_t duty, uint32_t clkDiv)**

There might be a better way of doing this.  However, it seems to keep the clock constant, which solves the "glitch" issue.  The PWM transitions smoothly now.

Best Regards

[pwm_lib.zip](https://github.com/antodom/pwm_lib/files/2941684/pwm_lib.zip)

antodom, 2019-03-12:
Hi there again,

I have made a small change to pwm_lib in order to avoid the clock change when using set_period_and_duty(). Now this function has a third argument, a boolean argument, keep_clock, for keeping the clock or not when you change the PWM period. By default keep_clock t is true, and it is a default argument, so with your actual code it should work without changes. 

The only thing is than you should verify that set_period_and_duty() returns true, if not, it would mean that the new period it is not possible with the clock the current clock.

Check if it works, and tell me.
Best.

Uup115, 2019-03-12:
Hello Antonio,

Yes, your changes seem to work well.  Thanks for your helpful code update!

I will keep my changes for only the start function:

**bool start (uint32_t period, uint32_t duty, uint32_t clkDiv)**

For my use case, the range of periods should never change.  Therefore, I don't anticipate the clock to change.  For this reason, I want to start with a "known" clock.  Your new code will return false, in the event that something goes wrong with the RPM. 

Thanks!

Best Regards

antodom, 2019-03-14:
Hi again Uup115,

I am glad the change was useful. In relation to starting with a given clock. A way to proceed without modifying the library  is to call function member start()  with the highest period of the range of possible periods (start() will select the clock with highest resolution for that period). As, now, function member set_period_and_duty(), by default, keeps the clock, and all the rest of periods in the range are smaller, it always will be successful (will return true) when called.

I hope it helps.
Best.

Uup115, 2019-03-20:
Hello Antonio,

Yes, your suggestion works well.  I will use your library without my changes. 

Thanks for your support!

Best Regards

closed 2019-03-20 by Uup115