antodom/pwm_lib #6: PWM won't stop?

charlesrwest, 2022-04-13:
Hello!

Thank you for making this library.  I'm hopefully going to be using it to control 12 half bridges driving 4 brushless motors.  So far it seems to be working fairly well with one exception.

I noticed that calling stop() on arduino_due::pwm_lib::pwm<arduino_due::pwm_lib::pwm_pin::PWMH1_PA19> Motor3EN2; didn't seem to do anything (it kept outputting).  I had to set the pin to a normal output and set it low to get it to stop.  

Thanks!

antodom, 2022-04-21:
Hi there,

There is no special treatment for this pin in pwm_lib, so I don't see why it is not working. This is the same pin used in pwm_lib's example basic_test.ino.

Well, to try to find out something, please, can you test your example but just with this pin working?

Best.

antodom, 2022-09-04:
I close this issue as there has been no answer since several months ago.

closed 2022-09-04 by antodom