antodom/pwm_lib #8: Inverted PWM?

erdavci, 2022-07-08:
In some cases we need an inverted PWM signal. How can we do this? Or better yet, can a function be added to the existing library to invert the PWM signal? (Like as PWM.start(period, duty, inverted = true [default: false])
Best Regards.

antodom, 2022-07-18:
Hi there @erdavci,

Sorry for the delay in my answer, but I have been very busy in the last weeks. Following your suggestion, I have modified function member start() for accepting a third flag argument to indicate that the output is inverted, which is false by default.

Please, update pwm_lib,  test it and tell me if it works correctly. 
I hope it helps.

erdavci, 2022-07-19:
[image]
I quickly tested it. It works perfectly. Thank you very much d^_^b

closed 2022-07-19 by erdavci